wabbitguy/Kobra_Max #4: higher parking position

marcuswitt, 2023-03-06:
would it be possible to make a higher park position ? 

it is always very tight when filament is dispensed to the print head during a changeover

wabbitguy, 2023-03-23:
@marcuswitt if you mean during the M600 filament change, the hot end at the change position is set to rise 20mm higher than the current print position. That's more or less a stock value for every printer I've ever seen or used. You need higher than that?

marcuswitt, 2023-03-24:
yes. that's what i mean. i have a bigger pile, unfortunately it's right on the corner of the bed where the output is.

wabbitguy, 2023-03-24:
@marcuswitt the problem stems from the fact that when the user selects "Pause" on the LCD, they usually want to do something to the objects on the bed, like insert a magnet or something. So the bed kicks forward, the hot end goes to the back left corner and sits an inch higher than the current print level. This works perfect for that purpose.

For a filament change, the same thing happens, but if you have a bed full of stuff, it can all get in the way when you're trying to change the filament and clean up the purge. Higher might help a little but now you're going to be fighting filament drool as the hot end slowly makes its way back to the point it left off at. Z axis moves on a bed slinger are glacial in speed.

I put a sheet of paper under my nozzle when it's doing a filament change and purge. It helps a little.

Changing the height of the park is going to bring in extra problems.  If you want a one off custom version for yourself with that one change, I can do that for you. Just let me know the height you want, and a way to contact you with a link for the download.

wabbitguy, 2023-03-24:
@marcuswitt just did some checking and all you need do is specify how/where you want the change to happen. In slicer where you normally insert the M600, add arguments to the M600 to override the default settings:

M600 X10 Y15 Z50 ; Do filament change at X:10, Y:15 and Z:+50 from current

or

M600 Z50 ; Do filament change at standard X, Y and Z:+50 from current

closed 2023-03-24 by wabbitguy