wabbitguy/Kobra_Max #5: Different thermistor defined in Cinfiguration.h than installed

kodi-ie, 2023-03-11:
I noticed that in firmware Configuration.h the thermistor #5 is defined. However (at least recently) there are a lot of Kobra Max with 3950 beta thermistor installed (#11). 

At this moment the difference in temperature is around 18% (i.e. displayed 190, real 215, displayed 230, real 275).

Would it be possible to make a build version with this thermistor configured?

wabbitguy, 2023-03-23:
@kodi-ie I've received replacement hot ends from Anycubic and the most the temp is out on my Fluke thermometer is 3C. I also tested a Kobra Max we just assembled and it's still within 4C.

Has Anycubic posted an official change in the thermistor used in the hot end? If they used a different thermistor from the onset I'd have no problem revising the firmware.

kodi-ie, 2023-03-23:
@wabbitguy, thank you very much for your answer. While there was no official information, there were reports of issues with the Kobra Max running too hot. A thread on Reddit discussed the issue, and multiple users have reported that their printer runs at least 20 degrees too hot (for example, see https://www.reddit.com/r/anycubic/comments/zrnm3l/comment/j15wxrn/?utm_source=share&utm_medium=web2x&context=3 and https://www.youtube.com/watch?v=EX51nvazbEY&t=231s&ab_channel=MandicReally). Based on my personal experience, I collected some data points and found that the thermistor readings corresponded to the 3950 curve in the Marlin source code. After changing the definition of the thermistor to #11 in Marlin, my Kobra Max was able to stabilize the hotend temperature within a few degrees of the displayed value.

Would you be so kind as to provide a build version for this thermistor? That way, I could direct others who are experiencing the same issue to try this firmware when they are unable to compile their own version.

Thank you very much for all your work!

wabbitguy, 2023-03-23:
@kodi-ie thanks for the follow up information! Very interesting indeed. I'll try the #11 thermistor setting on my Max, and if you want to try it as well, leave me a way I can contact you and I'll provide a compiled firmware beta link for you. I'd like to work this out before publishing it as the be all and end all solution...

wabbitguy, 2023-03-29:
@kodi-ie V1.6 is now released and includes the new thermistor setting (thanks for the heads up on it; works perfectly).

closed 2023-03-29 by wabbitguy