wabbitguy/Kobra_Max #6: Two problems after update

n1k0-83, 2023-03-29:
Thank you for this excellent firmware. 

Unfortunately, after installing it on two Kobra Max, I experienced the following problems:

1. when printing via USB (Repetier) the screen does not switch. 

2. when printing (SD or USB) the hotend shakes. It can neither draw a simple circle, nor a straight line without trembling. There are constantly minimal but very fast movements. 

I did the installation strictly according to the instructions. 

Any ideas?

Thanks and many greetings. 
Niko

wabbitguy, 2023-03-30:
@n1k0-83 we just tested extensively with Repetier Server Pro 1.48 and Octoprint. When starting a print job, the printer homes, the screen shows "Homing" message, when homing is done, the screen will change to the normal print screen and the LCD will show Host Printing Message. If you check the version on the LCD screen (About) it should say V1.6.

None of us can duplicate the shaking of the hot end when printing. You can try a sample GCODE we sliced to test with. If it doesn't cause any shaking then it's the slicer that's doing it. If it does, then it would have to be a mechanical issue I'd think.

[BookmarkEyes_0.2mm_PLA_16m.gcode.zip](https://github.com/wabbitguy/Kobra_Max/files/11107516/BookmarkEyes_0.2mm_PLA_16m.gcode.zip)

n1k0-83, 2023-03-31:
Just for understanding: Both KMax are absolute original. I've changed only the nozzles to a 0.6. Both are controlled by SD or Repetier. I'll test the update again today and provide you the results.

wabbitguy, 2023-03-31:
@n1k0-83 thanks for the clarification. Looking forward to anything you might find. I might also have one option I can disable in the firmware to see if that makes any difference on your Max but we'll cross that bridge based on your results.

closed 2023-04-04 by wabbitguy

n1k0-83, 2023-04-11:
[image]

n1k0-83, 2023-04-11:
Update Repetier Server Pro to version 1.4.10 has also not brought any improvement.

wabbitguy, 2023-04-11:
@n1k0-83 I'd need to see your printers config, so with a host connected, and when you're experiencing the vibrating hot end, do an M503 in the terminal, copy the info it reports and post that here.

n1k0-83, 2023-04-11:
@wabbitguy, thanks for your fast reply. Following the results of M503:

**KMAX 1**
Send:21:02:22.563: N15 M503
Recv:21:02:22.571: echo:  G21    ; Units in mm (mm)
Recv:21:02:22.574: echo:; Filament settings: Disabled
Recv:21:02:22.574: echo:  M200 S0 D1.75
Recv:21:02:22.577: echo:; Steps per unit:
Recv:21:02:22.580: echo: M92 X80.00 Y80.00 Z400.00 E405.00
Recv:21:02:22.583: echo:; Maximum feedrates (units/s):
Recv:21:02:22.589: echo:  M203 X200.00 Y200.00 Z40.00 E60.00
Recv:21:02:22.592: echo:; Maximum Acceleration (units/s2):
Recv:21:02:22.596: echo:  M201 X700.00 Y600.00 Z100.00 E3000.00
Recv:21:02:22.605: echo:; Acceleration (units/s2): P<print_accel> R<retract_accel> T<travel_accel>
Recv:21:02:22.608: echo:  M204 P800.00 R1000.00 T1000.00
Recv:21:02:22.620: echo:; Advanced: B<min_segment_time_us> S<min_feedrate> T<min_travel_feedrate> X<max_x_jerk> Y<max_y_jerk> Z<max_z_jerk> E<max_e_jerk>
Recv:21:02:22.626: echo:  M205 B20000.00 S0.00 T0.00 X0.00 Y5.00 Z0.30 E10.00
Recv:21:02:22.630: echo:; Home offset:
Recv:21:02:22.634: echo:  M206 X0.00 Y0.00 Z0.00
Recv:21:02:22.634: echo:; Auto Bed Leveling:
Recv:21:02:22.636: echo:  M420 S1 Z0.00
Recv:21:02:22.639: echo:  G29 W I0 J0 Z-0.90833
Recv:21:02:22.642: echo:  G29 W I1 J0 Z-0.77583
Recv:21:02:22.645: echo:  G29 W I2 J0 Z-0.58917
Recv:21:02:22.648: echo:  G29 W I3 J0 Z-0.59167
Recv:21:02:22.651: echo:  G29 W I4 J0 Z-0.70750
Recv:21:02:22.654: echo:  G29 W I5 J0 Z-0.89167
Recv:21:02:22.655: echo:  G29 W I6 J0 Z-0.81833
Recv:21:02:22.658: echo:  G29 W I0 J1 Z-1.03167
Recv:21:02:22.661: echo:  G29 W I1 J1 Z-0.74083
Recv:21:02:22.664: echo:  G29 W I2 J1 Z-0.68250
Recv:21:02:22.667: echo:  G29 W I3 J1 Z-0.66583
Recv:21:02:22.670: echo:  G29 W I4 J1 Z-0.69917
Recv:21:02:22.673: echo:  G29 W I5 J1 Z-0.84750
Recv:21:02:22.676: echo:  G29 W I6 J1 Z-0.85667
Recv:21:02:22.679: echo:  G29 W I0 J2 Z-0.88667
Recv:21:02:22.682: echo:  G29 W I1 J2 Z-0.70667
Recv:21:02:22.685: echo:  G29 W I2 J2 Z-0.52917
Recv:21:02:22.685: echo:  G29 W I3 J2 Z-0.48333
Recv:21:02:22.688: echo:  G29 W I4 J2 Z-0.46583
Recv:21:02:22.693: echo:  G29 W I5 J2 Z-0.48333
Recv:21:02:22.695: echo:  G29 W I6 J2 Z-0.47167
Recv:21:02:22.698: echo:  G29 W I0 J3 Z-0.54667
Recv:21:02:22.701: echo:  G29 W I1 J3 Z-0.34167
Recv:21:02:22.704: echo:  G29 W I2 J3 Z-0.32750
Recv:21:02:22.707: echo:  G29 W I3 J3 Z-0.32833
Recv:21:02:22.710: echo:  G29 W I4 J3 Z-0.35500
Recv:21:02:22.713: echo:  G29 W I5 J3 Z-0.45167
Recv:21:02:22.716: echo:  G29 W I6 J3 Z-0.52250
Recv:21:02:22.716: echo:  G29 W I0 J4 Z-0.69000
Recv:21:02:22.719: echo:  G29 W I1 J4 Z-0.49000
Recv:21:02:22.723: echo:  G29 W I2 J4 Z-0.19583
Recv:21:02:22.726: echo:  G29 W I3 J4 Z-0.02750
Recv:21:02:22.729: echo:  G29 W I4 J4 Z0.06583
Recv:21:02:22.732: echo:  G29 W I5 J4 Z0.12000
Recv:21:02:22.735: echo:  G29 W I6 J4 Z0.15750
Recv:21:02:22.738: echo:  G29 W I0 J5 Z0.20000
Recv:21:02:22.741: echo:  G29 W I1 J5 Z0.19583
Recv:21:02:22.741: echo:  G29 W I2 J5 Z-0.01917
Recv:21:02:22.744: echo:  G29 W I3 J5 Z0.12417
Recv:21:02:22.747: echo:  G29 W I4 J5 Z0.10000
Recv:21:02:22.750: echo:  G29 W I5 J5 Z0.01167
Recv:21:02:22.753: echo:  G29 W I6 J5 Z-0.02917
Recv:21:02:22.757: echo:  G29 W I0 J6 Z-0.18417
Recv:21:02:22.760: echo:  G29 W I1 J6 Z-0.14833
Recv:21:02:22.763: echo:  G29 W I2 J6 Z-0.01833
Recv:21:02:22.766: echo:  G29 W I3 J6 Z-0.29500
Recv:21:02:22.766: echo:  G29 W I4 J6 Z-0.08333
Recv:21:02:22.769: echo:  G29 W I5 J6 Z-0.18083
Recv:21:02:22.772: echo:  G29 W I6 J6 Z-0.03667
[... 86 lines cut ...]
Recv:21:08:11.926: echo:  G29 W I4 J6 Z-1.75583
Recv:21:08:11.929: echo:  G29 W I5 J6 Z-1.84333
Recv:21:08:11.932: echo:  G29 W I6 J6 Z-1.94083
Recv:21:08:11.933: echo:; PID settings:
Recv:21:08:11.936: echo:  M301 P22.20 I1.08 D119.00
Recv:21:08:11.939: echo:  M304 P97.10 I1.41 D1675.16
Recv:21:08:11.942: ; Controller Fan
Recv:21:08:11.945: echo:  M710 S255 I0 A1 D60 ; (100% 0%)
Recv:21:08:11.948: echo:; Power-Loss Recovery:
Recv:21:08:11.948: echo:  M413 S1
Recv:21:08:11.951: echo:; Z-Probe Offset (mm):
Recv:21:08:11.954: echo:  M851 X0 Y0 Z0.05
Recv:21:08:11.957: echo:; Stepper driver current:
Recv:21:08:11.960: echo:  M906 X800 Y1100 Z1300
Recv:21:08:11.963: echo:  M906 T0 E800
Recv:21:08:11.963: echo:; Driver stepping mode:
Recv:21:08:11.966: echo:  M569 S1 X Y Z
Recv:21:08:11.970: echo:; Filament load/unload lengths:
Recv:21:08:11.972: echo:  M603 L550.00 U700.00
Recv:21:08:11.976: echo:; Filament runout sensor:
Recv:21:08:11.979: echo:  M412 S1 D15.00
Recv:21:08:11.979: ok

Best regards
Niko

n1k0-83, 2023-04-11:
Here while travelled with the issue:

Exec:21:11:10.937: @moveAbs X404.00
Send:21:11:10.938: N14 G1 X404.00 F6000
Exec:21:11:10.938: @updatePrinterState
Recv:21:11:17.015: echo:busy: processing (3)
Recv:21:11:17.458: ok
Send:21:11:21.399: N15 M503
Recv:21:11:21.406: echo:  G21    ; Units in mm (mm)
Recv:21:11:21.409: echo:; Filament settings: Disabled
Recv:21:11:21.409: echo:  M200 S0 D1.75
Recv:21:11:21.412: echo:; Steps per unit:
Recv:21:11:21.416: echo: M92 X80.00 Y80.00 Z400.00 E405.00
Recv:21:11:21.419: echo:; Maximum feedrates (units/s):
Recv:21:11:21.425: echo:  M203 X200.00 Y200.00 Z40.00 E60.00
Recv:21:11:21.428: echo:; Maximum Acceleration (units/s2):
Recv:21:11:21.431: echo:  M201 X700.00 Y600.00 Z100.00 E3000.00
Recv:21:11:21.440: echo:; Acceleration (units/s2): P<print_accel> R<retract_accel> T<travel_accel>
Recv:21:11:21.444: echo:  M204 P800.00 R1000.00 T1000.00
Recv:21:11:21.456: echo:; Advanced: B<min_segment_time_us> S<min_feedrate> T<min_travel_feedrate> X<max_x_jerk> Y<max_y_jerk> Z<max_z_jerk> E<max_e_jerk>
Recv:21:11:21.462: echo:  M205 B20000.00 S0.00 T0.00 X0.00 Y5.00 Z0.30 E10.00
Recv:21:11:21.465: echo:; Home offset:
Recv:21:11:21.468: echo:  M206 X0.00 Y0.00 Z0.00
Recv:21:11:21.468: echo:; Auto Bed Leveling:
Recv:21:11:21.471: echo:  M420 S1 Z0.00
Recv:21:11:21.474: echo:  G29 W I0 J0 Z-0.90833
Recv:21:11:21.478: echo:  G29 W I1 J0 Z-0.77583
Recv:21:11:21.481: echo:  G29 W I2 J0 Z-0.58917
Recv:21:11:21.484: echo:  G29 W I3 J0 Z-0.59167
Recv:21:11:21.487: echo:  G29 W I4 J0 Z-0.70750
Recv:21:11:21.490: echo:  G29 W I5 J0 Z-0.89167
Recv:21:11:21.490: echo:  G29 W I6 J0 Z-0.81833
Recv:21:11:21.493: echo:  G29 W I0 J1 Z-1.03167
Recv:21:11:21.496: echo:  G29 W I1 J1 Z-0.74083
Recv:21:11:21.499: echo:  G29 W I2 J1 Z-0.68250
Recv:21:11:21.502: echo:  G29 W I3 J1 Z-0.66583
Recv:21:11:21.506: echo:  G29 W I4 J1 Z-0.69917
Recv:21:11:21.509: echo:  G29 W I5 J1 Z-0.84750
Recv:21:11:21.512: echo:  G29 W I6 J1 Z-0.85667
Recv:21:11:21.516: echo:  G29 W I0 J2 Z-0.88667
Recv:21:11:21.518: echo:  G29 W I1 J2 Z-0.70667
Recv:21:11:21.521: echo:  G29 W I2 J2 Z-0.52917
Recv:21:11:21.521: echo:  G29 W I3 J2 Z-0.48333
Recv:21:11:21.524: echo:  G29 W I4 J2 Z-0.46583
Recv:21:11:21.527: echo:  G29 W I5 J2 Z-0.48333
Recv:21:11:21.530: echo:  G29 W I6 J2 Z-0.47167
Recv:21:11:21.534: echo:  G29 W I0 J3 Z-0.54667
Recv:21:11:21.537: echo:  G29 W I1 J3 Z-0.34167
Recv:21:11:21.540: echo:  G29 W I2 J3 Z-0.32750
Recv:21:11:21.543: echo:  G29 W I3 J3 Z-0.32833
Recv:21:11:21.546: echo:  G29 W I4 J3 Z-0.35500
Recv:21:11:21.549: echo:  G29 W I5 J3 Z-0.45167
Recv:21:11:21.552: echo:  G29 W I6 J3 Z-0.52250
Recv:21:11:21.552: echo:  G29 W I0 J4 Z-0.69000
Recv:21:11:21.555: echo:  G29 W I1 J4 Z-0.49000
Recv:21:11:21.558: echo:  G29 W I2 J4 Z-0.19583
Recv:21:11:21.561: echo:  G29 W I3 J4 Z-0.02750
Recv:21:11:21.564: echo:  G29 W I4 J4 Z0.06583
Recv:21:11:21.568: echo:  G29 W I5 J4 Z0.12000
Recv:21:11:21.571: echo:  G29 W I6 J4 Z0.15750
Recv:21:11:21.574: echo:  G29 W I0 J5 Z0.20000
Recv:21:11:21.577: echo:  G29 W I1 J5 Z0.19583
Recv:21:11:21.577: echo:  G29 W I2 J5 Z-0.01917
Recv:21:11:21.580: echo:  G29 W I3 J5 Z0.12417
Recv:21:11:21.583: echo:  G29 W I4 J5 Z0.10000
Recv:21:11:21.586: echo:  G29 W I5 J5 Z0.01167
Recv:21:11:21.589: echo:  G29 W I6 J5 Z-0.02917
Recv:21:11:21.592: echo:  G29 W I0 J6 Z-0.18417
Recv:21:11:21.595: echo:  G29 W I1 J6 Z-0.14833
Recv:21:11:21.599: echo:  G29 W I2 J6 Z-0.01833
Recv:21:11:21.602: echo:  G29 W I3 J6 Z-0.29500
Recv:21:11:21.602: echo:  G29 W I4 J6 Z-0.08333
Recv:21:11:21.605: echo:  G29 W I5 J6 Z-0.18083
Recv:21:11:21.608: echo:  G29 W I6 J6 Z-0.03667
Recv:21:11:21.611: echo:; PID settings:
Recv:21:11:21.614: echo:  M301 P22.20 I1.08 D119.00
Recv:21:11:21.617: echo:  M304 P97.10 I1.41 D1675.16
Recv:21:11:21.622: ; Controller Fan
Recv:21:11:21.623: echo:  M710 S255 I0 A1 D60 ; (100% 0%)
Recv:21:11:21.626: echo:; Power-Loss Recovery:
Recv:21:11:21.627: echo:  M413 S1
Recv:21:11:21.630: echo:; Z-Probe Offset (mm):
Recv:21:11:21.633: echo:  M851 X0 Y0 Z0.05
Recv:21:11:21.636: echo:; Stepper driver current:
Recv:21:11:21.639: echo:  M906 X800 Y1100 Z1300
Recv:21:11:21.639: echo:  M906 T0 E800
Recv:21:11:21.642: echo:; Driver stepping mode:
Recv:21:11:21.645: echo:  M569 S1 X Y Z
Recv:21:11:21.648: echo:; Filament load/unload lengths:
Recv:21:11:21.651: echo:  M603 L550.00 U700.00
Recv:21:11:21.654: echo:; Filament runout sensor:
Recv:21:11:21.656: echo:  M412 S1 D15.00
Recv:21:11:21.657: ok

wabbitguy, 2023-04-11:
@n1k0-83  okay I know exactly what the problem is...I've seen it once before on one of the test Max printers.

From your host terminal, send these two lines to fix it:

M205 X5
M500

If you also are using a stock extruder, the values for the extrusion are out about 5%.. you can adjust those if you want to go through it all with extruding and testing or just put in:

M92 E420
M500

n1k0-83, 2023-04-12:
Thank you so much!
The shaking problem no longer exists. The movements are smooth. 

[image]

n1k0-83, 2023-04-12:
@wabbitguy So all issues are fixed. Both KMax running without problems and the print results are quit perfect. Even with the 0.6 nozzle, the results are simply great. Thank you so much for your great firmware and your awesome support. You made my day!

wabbitguy, 2023-04-12:
@n1k0-83 you're very welcome, great to hear it's working well for you. I tried a 0.6 nozzle but had issues with stringing that I could never solve, so it's also good to hear you got that working.

n1k0-83, 2023-04-12:
Both KMax works with wonderful with your firmware and 0.4 and the 0.6 nozzle. My setup is most of time 0.6 @ KMax and 0.4 @ Vyper. Btw. hope that I can find your firmware for the Vyper, too. It would be great ;-)

wabbitguy, 2023-04-13:
@n1k0-83 I've never looked at the firmware on the Vyper, I did do some for the Kobra Plus and the Chiron. The Vyper looks like it uses the same display as the Kobra series, no idea what motherboard they use though and without one in the shop here to test on it's a big shot in the dark...no idea when the current version of the Vyper firmware is either.

wabbitguy, 2023-04-13:
@n1k0-83 just to be clear, if I did work on the Vyper firmware there would need to be some discussion. Like is a 7x7 levelling grid beneficial to the Vyper? Whats the total length of the Bowden tube in mm? And some other things that may arise.

Secondly, I would need a beta tester willing to install the firmware and screen update on their Vyper. Prior to that they would need access to the original Anycubic Vyper firmware for the printer AND LCD so in the event something goes amiss, their printer can be restored to factory specification. 

Lastly the firmware update for the Vyper could or may wipe out any previous settings so possibly any levelling, PID tuning, Z-offset, e-Steps would need to be redone. Pronterface via USB (or some host) would be required and the beta tester would need knowledge of how to do all these things (send and receive GCODES)..etc

n1k0-83, 2023-04-13:
I see no problem and would like to support you. However, I use the Community firmware on the Vyper, which also includes PID tuning in the scope of functions. Even if you write in the other post that this is useless, so I would not want to do without this. If you would be available for this deal, I would really like to support you. You can take a look at the community firmware as an example and for comparison: https://github.com/CR6Community

wabbitguy, 2023-04-13:
@n1k0-83 sorry but I just can't see the logic in PID tuning.

PID tuning doesn't even take effect until 10C below the target temp since it basically designed to keep the temp within a range at target temp. If the Vyper is anything like the Max, Anycubic has the wrong thermistor defined in the configuration as well. So applying PID tuning for the wrong partakes even less sense.

I do appreciate the fact that my Max firmware has enabled you to enjoy your printer more fully.