wabbitguy/Kobra_Max #7: I am having issues with measuring correct Hot End temperatures

Karny14, 2023-04-04:
My Kobra Max Hot end is off by atleast 20-30 degrees Celsius. Is this my error by not using pronterface and using commands 502 & 500?

Or is my measuring device incorrect

wabbitguy, 2023-04-04:
@Karny14 when flashing the firmware without using the M502/M500, you are using the old values from the factory firmware (if that was the last firmware used in the printer). The thermistor will be correct, but a lot of the other information will be ignored or completely wrong. Like the bed mesh. The factory firmware uses a 25 point bed levelling system, the new firmware uses a 49 point which superior for a bed the size of the Max.

The factory firmware has the wrong thermistor selected in the printer and that was addressed and corrected with the release of V1.6.

As for measuring the temperature of the hot end, the only accurate way is with a calibrated temperature monitor and having the sensor installed in the hot end heater block the same as the stock version is. Thats how we tested them.

For example, if you put a temp probe in the sock of the heater block and it reads the same temp as the currently set temperature that's incorrect right off. There should be a 15 to 30 degree difference from real thermistor location to anywhere on the outside of the heater block.

Karny14, 2023-04-04:
Hello, thanks for the quick reply.

I have flashed the new firmware and followed your steps to use pronterface and semd gcode commands 502 & 500.

The bed levelling is working great no issues with that.

Prints are manageable, but will most likely need to figure out my temperature values as I have been getting crazy stringing on my prints.

Could possibly be too hot of a hot end or not correct retraction settings. I believe I am running 6mm retraction, 40 mm/s retract and prime speed. With 0.048mm3 for extra prime amount. 

For the most part print is great and functional, just trying to get the premium quality I have on my other anycubic printer (Mega S)

Thanks again for all your help, you are amazing for helping me out and everyone else 🙂

wabbitguy, 2023-04-04:
@Karny14 stringing is a pain thats for sure. I converted one of the Max's to a 0.6mm nozzle and it put new meaning to word drool. Never could get it tamed, did everything from retraction speed/length, z hop, temperatures, print speeds and more.

The best we could get was using PRUSA Slicer's volumetric slope settings. A setting of 3 in the positive and negative helped a lot. But changing back to a 0.4 nozzle and it was gone.

For a .4 nozzle, retraction speed of 40 with a length of 6mm worked best here but no extra prime. And some filament brands will string no matter we do. Silk PLA's for example. Stringy in their nature. Fortunately here we have low humidity so moisture has never been an issue.

The Max probably won't hold a candle to the Mega S. The MEGA S was one of these printers Anycubic ever turned out. But the Max is superior to the Chiron's we have here.

Hope you get it sorted out!

Karny14, 2023-04-04:
I guess you solved my issue haha, i changed to a 0.6 mm nozzle in the past two months and the retraction has just been a phenomenom that I cant figure out. Thanks again, seems like my problems are all man made. Thanks again, the super accurate bed levelling is awesome and all the other changes, fillament removal and changing. Plus the micro z offset steps have made my printer much better!

Thanks for all the support.

wabbitguy, 2023-04-04:
@Karny14 LOL...if you ever get that 0.6 nozzle figured out, be sure to let us know!!!!!

closed 2023-04-04 by wabbitguy