wabbitguy/Kobra_Max #8: motor noise

rioaje, 2023-04-04:
after updating the firmware using this one, the motor is making a noise at idle.

i tried switch back to original firmware again and the noise is gone. what is the cause and which setting is causing this?

wabbitguy, 2023-04-09:
@rioaje The firmware settings in the 1.6 release are identical to the stock settings except for the current on the Y axis which was increased to stop Y axis shifts.

If you use the "Control" button from the main screen and select "Motors Off", the steppers will disengage the motors. If that stops the noise, then it's the motor(s) holding their position. Which eventually times out and they shut off themselves any way.

rioaje, 2023-04-09:
Yes the motor noise is gone after i select motors off. The motor is always making noise whenever the motor is on (during leveling), which is weird. I tried rebuild the firmware and reduce the motor current, it still making the same noise. That a bit weird.

So in the end i swapping back to original firmware

wabbitguy, 2023-04-11:
@rioaje in the ConfigurationADV.h look for #define SQUARE_WAVE_STEPPING and change it to:

//#define SQUARE_WAVE_STEPPING

That will disable it. Makes the motors get a sloppy pulse from the motherboard but the it might fix a borderline stepper driver and kill the noise. Its the only difference between stock and V1.6.

n1k0-83, 2023-04-13:
Finally everything worked without problems and now... 
After a day without problems, today when I print for the first time, I also have these problems.

I notice that these issues occur for short periods of time after the first layer.
At the moment I am in the sixth layer and nothing more can be "heard" of the issue.

If you think that this noise does not have a negative impact on the hardware, I would not change anything for now.

wabbitguy, 2023-04-13:
@n1k0-83 if the noise comes and goes, it's probably from a stepper motor that's hitting the resonant frequency of its mounting setup. In other words, it vibrates through the frame of the printer itself. If the Z rods and the bed are in a certain position/area the motor noise can use and make it audible to us. That's pretty common in these rather cheaply built printers.

It's certainly not going to hurt any hardware, it's just the frame/stepper/mount is acting like a big speaker. Varying the speed of the printer with the slicer can make the noise louder or more quiet. And that's pretty common as well.

closed 2023-04-13 by wabbitguy