wabbitguy/Kobra_Max #10: Firmware stuck in print screen after USB print

wineds, 2023-04-22:
I am having issues with the Firmware stuck in print screen after USB print.  This could be printing via USB from Prontoface or Cura or Octoprint.  Can't recall if this was also an issue with stock FW.  Could this be related to having to tape the +5v on SUB cable to printer?

Great job on the FW by the way.  Now the HE temps are correct and can finally set Z offset to correct value.

wabbitguy, 2023-04-26:
@wineds firmware was tested by multiple users with Octoprint (1.8.7), Repetier Server Pro(1.4.9) and no one can duplicate your issue. All users with Raspberry PI's did have the 5V pin taped over, the others were direct from Mac or Windows with regular USB cable. Tested with Pronterface (aka PrintRun 2.0.0rc8).

No one tested printing via CURA, none of the testers use it.

Stock firmware never changes to the print screen, it remains at the main screen display.

Perhaps the end GCODE isn't triggering the firmware (PRUSA slicer end geocode used):

`M104 S0  ;extruder heater off
M140 S0  ;heated bed heater off (if you have it)
G91      ;relative positioning
G1 Z+10 F3600 ;move Z up a bit
G90              ;absolute positioning
G1 X10 F3000    ; get the head off the bed
G1 F3000 Y400    ;kick the bed out
M84        ;steppers off
M355 S0    ;turn off the case light`

n1k0-83, 2023-04-27:
I just want to mention that I have tested all possibilities, including various test models and Gcode adjustments, the firmware with Cura 5.2.2 and 5.3. On my two KMax everything works without problems and restrictions. Many greetings.

wineds, 2023-04-27:
Thanks @wabbitguy   Issue seems to be resolved after update to 1.8.  I do notice that you need to wait until after the final beep at end of print before pressing confirm on the touch screen?  Otherwise I still get locked out and have to power cycle.

wabbitguy, 2023-04-27:
@wineds the only place there are "beep" sounds are:

-  in filament runout
- a confirm if the printer is paused too long and the heater times out
- when bed level probing fails
- resuming from a power loss
- at power on

At no other time is there sound. So whatever yours is doing is unique to either your Octoprint config or a problem directly with the motherboard itself.

closed 2023-04-27 by wabbitguy