wabbitguy/Kobra_Max #13: axis shaking

djepsylon, 2023-05-10:
hi,
nice job for this firmware! 
i ve see a little bug with your fw, after few days, X axis and Y axis shaking.
when i put another firmware it works well and if i put your firmware it works too but after few days axes starts shaking i don't know why.
this is only with this release.
have you an idea to resolve this? 
this is boring to flash the printer every weeks lolll.

wabbitguy, 2023-05-16:
@djepsylon the video doesn't look like any Kobra Max I've ever seen. The display certainly isn't a max display. So if someone else's firmware makes whatever printer that is run smoothly, I'd use it. Trying to trouble shoot an issue that appears after a few days would be impossible, could be anything from simple stepper driver over heating to some value in the EPROM memory being reset (checking with M503 would address that).

closed 2023-05-16 by wabbitguy

djepsylon, 2023-05-21:
> @djepsylon the video doesn't look like any Kobra Max I've ever seen. The display certainly isn't a max display. So if someone else's firmware makes whatever printer that is run smoothly, I'd use it. [...]

sorry this is just a missclick when i uploaded the video

wabbitguy, 2023-05-21:
@djepsylon so this video link says I need access, the last one didn't. I don't have the luxury of time to toy with GoggleDrive.

The M503 might provide a hint, apart from that if it happens after a few days, it's still impossible to determine what the issue could be.

Too many things it "could" be. i.e. after a "few" days could be a print you're running for 48hrs or longer, in which case a totally different set of problems, all associated with the motherboard more than anything else. If you're doing a 1hr print each day and it shakes after a few days, again, where does one even start. Same GCODE? Different GCODE? CURA Slicer? PRUSA Slicer? PETG, vs TPU/ASA/HIPS/PLA? Printer via SD? Octoprint? Repetier Server? And what other firmware version works, whereas the 1.8.1 doesn't? And just flashing or starting from scratch each time (with level/PID/esteps/etc)...

And probably 20 or 30 more questions to narrow it down, and then another bunch of questions when and if it can be narrowed down at all.

djepsylon, 2023-05-26:
hi, sorry for lost of time but me to i ve not really time to play with unutiles things. i have my own society and time is not enough for averyone. well i ve missed to add permission when i maked the link sorry. you must see the video beacause a lot of poeplo where i did gave your firmware have this problem and a problem with leveling. the nozzle want tou pass trough the bed or push too much on it
 if you have an idea to tri i can recompile on firmware with parameters you want i change

https://drive.google.com/file/d/1eYY0C9W9F8k4ycXncFOHknnCzzM29pXE/view?usp=sharing

wabbitguy, 2023-05-26:
That looks exactly like the issue from the error in version 1.8 and earlier versions including 1.6,  that I found and corrected in version 1.8.1. Hence 1.8.1 is the only stable release that should be used, but any wrong values from previous versions will still be stored in EPROM.

Thanks VERY much for the updated link! Very easy to see the problem now. And I recognize it as well since it has happened to me.

Reflashing the firmware alone will not fix anything. The bug will remain in EPROM memory until it is fixed via Pronterface (or some other terminal host) with GCODE or completely flashed with 1.8.1 and then with a factory reset (M502, M500) which will require full bed levelling and eSteps again.

When running Pronterface, issue an M503 command from its terminal panel. You'll get back a wealth of information but you are looking for this line:

echo: M205 B20000.00 S0.00 T0.00 X5.00 Y5.00 Z0.30 E10.00

With the shaking, the X or Y value will be 0 (zero) and it should be 5.00 (the error was the X value was set to 0.00)

To fix it, again from the terminal, type in (the Y value is merely a precaution):

M205 X5.00 Y5.00
M500

That will reset the value to what it should have been and save it in EPROM memory, no flashing or updating is required.

For the levelling errors, that's not a firmware issue in any case. Mine also pushes the nozzle through the bed in three places and that's happened originally with the Anycubic original firmware. The problem, near as we can figure out is the strain gauge. If there any wires pushing on the hot end inside the shroud they will influence the strain gauge.

The bowden tube is about 20mm shorter than it should be as well and in some positions the tube will pull the hot end and not allow it to detect properly. We now remove the Bowden tube from top of the hot end when we do levelling and it has helped a great deal.

Before levelling, after the Bowden is removed, look down through the top of the shroud and there should be no LED lit on the left side of the hot end. If it is something is putting pressure against the hot end and triggering the strain gauge (hence the reason the LED is on).

For the firmware to toggle to the Print Screen when printing from a host, that's all done in the DGUS module. It's in the page1_handle() section near the bottom of the function:

`			if (printer_state != AC_printer_idle && isPrinting()) {
			  gcodeComment = "Host Printing Mode";// MEL_MOD indicate where GCODE coming from
        ChangePageOfTFT(PAGE_STATUS2);// MEL_MOD auto change to PRINT page if printing via USB
      } `

This is the main screen display function that is looped through waiting for a button press or something to happen. What I check for is if the printer is NOT idle and has the "isPrinting" flag set I know it's printing from the USB port so I flip the Page Status to PAGE 2 which is the Print screen. If you wanted other printers to do that, you'd have to add in the code to do similar checks and them move to a print screen.

Anycubic used the board and CPU because they probably had a bunch of them made at a low cost. These companies are all about money money and very little about customer satisfaction. Even at that there's a wiring error on the board and linear advance will not allow independent control the extruder (stealthchop to spreadCycle). A resistor has to be moved to the correct position and then the firmware for UART control has to be updated to use it.

I suspect AnyCubic used KeilARM because the company they hired used it and there's little support for the CPU in other compilers so we're kind of stuck with it. I know a lot of people have taken out the Tri-Gorilla board and used something else just so they could control the printer properly.

We tried to redo the firmware so we could use a BLTouch instead of the strain gauge. That turned out to be impossible because AnyCubic completely erased all the contents of the Servo libraries. For some reason known only to them. I've tried putting them back in but there's errors either from KeilArm or something else I'm missing and I've given up. Probably be easier to use a nozzle probe.

Electrical wise, the Kobra is okay. Noisy fans, sub-standard tinned wires for connecting to the MOSFETS for the bed, stepper drives soldered to the motherboard so if one blows out the whole board goes in the garbage. The USB to Serial chip (CH340) is a piece of junk and I wish they'd used something with more quality. We get USB drop outs on 3 of the Max's I use for testing. But no doubt for AnyCubic is was the cheapest way to go.

Mechanical, it's quite sturdy but it would have been even better if they had used linear rails for the Y axis instead of some kids roller skate wheels...LOL..it would have made Y axis layer shifts maybe happen less as well. I find when doing many hour large prints at too high a speed, I get layer shifts. No amount of jerk/accel settings fix it, so I stay 60mm/s or slower.

Any way, let me know if that M204 fixes the jittering error.

Mel
---

djepsylon, 2023-05-27:
hi @wabbitguy 
thank you for your reply, i did the m205 and it's ok it's corrected. i don't understand why i didn't see it before i did m503 hundreds of times in the terminal of my octoprint!! i must not have had my eyes in front of the holes LOL.
I'm thinking of testing for a few more days with the original card and then, if I don't get anywhere, I think I'll opt for one of the following two solutions:
-either correct the wiring error and unsolder the shunt and replace it correctly on the motherboard for the extruder motor driver and put it under klipper.
-Replace the motherboard with a Ruby card from artillery that I have in stock with tmc 2209s and x2 firmware modified with marlin 2.1.2 and the input shaper.

I'm leaning more towards the 2nd solution as I don't think klipper is 100% reliable yet and for certain things, marlin isn't as fast, admittedly, but it's much better quality. a simple example is the management of overangles and bridges which are much better printed with marlin than with klipper and I think that input shaping could be a very useful function on the kobra max.

wabbitguy, 2023-05-27:
@djepsylon great to hear you got it working! I apologize for the error, I've poured over my source code a number of times and experienced the same X axis issue but just couldn't see why it was doing it. I finally had some help from another and found it in a module I don't even remember editing. Probably a senior moment...

The DGUS code is a nightmare. There's no other way to put it. The display is closed source firmware that merely displays a bunch of BMP's when told to do so. There's a DGUS editor you can use to define your own displays (for controls and text display) then create a config file for it and flash the display. But really the display is still mostly brain dead because Marlin doesn't send all the data needed to it so it knows what's going on. If there was a normal LCD slot pair on the mother board we could at least put a usable read out on the printer but, it's nothing more than a serial port.

I've seen a few guy tell me they are running Klipper on the Max. When I ask for details things get very quiet and I rarely hear back from them. And when I do its links to Klipper with no details at all. Nor have I seen one printed item from any of the Max Klipper users. So its not that I'm skeptical, I just need to see to be sure of anything...

If I were going to redo the Kobra Max, I'd take that motherboard out of it and throw it away. Yes, there'd be a pile of wiring to do for a new board, but I'd also put on a decent display that would give complete control of all Marlins features. If my motherboard burns out, that's what I'd do. As you point out, Klipper is probably nice for speed, but speed isn't something that's all important to me. Reliability and versatility is.

I've toyed with a 0.6 and 0.8 nozzle on the Max. The 0.8 is like a Saint Bernard on a hot day. Retraction and drooling is really hard to contain. A 0.6 is a little better, but on large prints, still drool worthy...LOL

As for the DGUS editor, I use DGUS_V7.641 if you need a bit of a quick walk thru I could probably help out a bit there. Once you do it, it's not bad. Problem is if you want to define NEW items, then you need to write it all down by hand in the editor, then put it all in the AnyCubic DGUS_xxx modules so it can access it. So it's not exactly a walk in the part. Well, maybe a walk in the dog park. Barefoot.

Mel

djepsylon, 2023-05-28:
thanks for you replies.
i love the st bernard's picture lolllll
i realy think i m going to put the ruby board in there without screen, i use octoprint, screen is unutile.
impossible to have good prints. quality with stock card is acceptable for tests or non visible prototype but not more, i want more in a 500€ printer! loll
thanks for your help. if you want, i can keep you informed when i finish tests after put ruby card

wabbitguy, 2023-05-28:
@djepsylon I used to run Octoprint but I have three deltas and Octoprint is far to slow for them so I had to hunt around for something faster. Ended up with Repetier Server Pro, not free like Octoprint but it can run all three deltas at the same time from a single Raspberry Pi the push a message to my phone/watch to let me know the print is done (or even has a problem).

However even with Repetier Server Pro or Octoprint on any of my Anycubic printers I can get communication errors and thus ends the print in failure. I think largely because of the CH340x series USB to serial chips used on those printers (CP2102/4 are 100 times more reliable but not a clone of anything like the CH340 series so too expensive to be considered would be my guess).

I'd be very interested to know how your update comes out and would appreciate that a lot! Just to find out what issues you run into would be extremely valuable information. Thank you very much for the offer!

Mel