wabbitguy/Kobra_Max #18: No VS compiler?

zonalimitatore, 2023-07-11:
This firmware can't be compiled under platformio?

maikramer, 2023-07-19:
You can wait until i finish my work on having a updated Marlin for the Kobra Max. I'm almost there.
You can build it with platformio and vs code.
https://github.com/maikramer/MarlinKobraMax
I still working on it, but the only things are not working are the runout sensor and the Power Loss recovery.
With my build you can use the updated Linear Advance and the much improved Input Shapping.

closed 2023-07-28 by wabbitguy

wabbitguy, 2023-07-29:
@zonalimitatore Anycubic didn't use any standard compiler so we were restricted to what we can use. It is either KeilARM compiler or nothing. In order to make it compatible with the more common compilers and even make it more Marlin friendly takes a lot of work to do. More than the printer is worthy of for me.

Mel