wabbitguy/Kobra_Max #21: Any chance you'll build this for the TJC screens?

TTVert, 2023-07-28:
Seems some of us less fortunate ones  have TJC screens, not DWIN screens so we cannot update the screen FW.  :(

Dave

wabbitguy, 2023-07-29:
@TTVert at present not a chance. Anycubic hasn't released any information about how the TJC screens require to be supplied with data, what apps can be used to build the data. Took us long enough to figure out the original DWIN nightmare.

Sorry I can't help.

Mel

closed 2023-07-29 by wabbitguy

TTVert, 2023-07-29:
> @TTVert at present not a chance. Anycubic hasn't released any information about how the TJC screens require to be supplied with data, what apps can be used to build the data. Took us long enough to figure out the original DWIN nightmare.
> 
> Sorry I can't help.
> 
> Mel

No worries I do appreciate the response. I need to see if I can find a dwin screen for my printer as I know that made them at some point.

Dave

wabbitguy, 2023-07-29:
@TTVert yeah it's a pain in the butt with AnyCubic. The run out of stuff and substitute whatever and never update any information about it. The original Tri-Gorilla boards apparently are gone and they are shipping out boards that have the resistor fixed (for the UART control of the steppers so linear advance might actually work right), and there's apparently new firmware for these boards but only compiled versions. Absolutely nothing on Anycubics GitHub site about the changes or updates.

I just learned about the new screens this week, before that I had no idea. But a few had trouble updating their screens and now I'll bet that's exactly what the issues were. Sigh.

I know when my Kobra Max needs a motherboard I'll be replacing it with something I can control and with a display I want to use and maintain.

Sorry about the delays in the response as well, for some unknown reason I don't always get notices from GitHub that there is an issue open. Add that to list of oddities...

Mel

wabbitguy, 2023-07-29:
@TTVert if you happen to come across any info on the new screens, like how to code for them, I'd really appreciate knowing.

Mel

TTVert, 2023-07-29:
> 

I think that may be a bit above my paygrade but in my research I did find [this](https://unofficialnextion.com/t/nextion-and-tjc-whats-the-difference/20).   They mention tools for cross compatibility but the tools were in Chinese.  Someone found a workaround to display in English but also elaborates much further on the differences between the two.  I had read somewhere that there may be insufficient storage to even modify or add images to the TJC screens.  With that I am trying to source a DWIN screen presently.  Do you know if they are capacitive or resistive screens?

Thanks
Dave

wabbitguy, 2023-07-29:
@TTVert absolutely awesome find. I search and managed to find more rabbit holes than information.

I agree with you on the new printers coming with the TJC screens. It's a saving in money and would make perfect sense.

I'll have a look at the editors and see if I can make some sense of it when I get a free moment.

Again, thanks for all your work and time Dave. Very much appreciated!

Mel

TTVert, 2023-07-29:
> @TTVert absolutely awesome find. I search and managed to find more rabbit holes than information.
>
> I agree with you on the new printers coming with the TJC screens. It's a saving in money and would [...]

I've not done nearly as much as you have done but I'm happy to help where I can.  I may also play w/ the TJC software when I get a chance.   I believe there is a much newer version of that software but they may have changed some things to not allow it to install on a win10 machine but I only briefly skimmed over that part.   The link I provided definitely works in win10.

Dave

TTVert, 2023-07-29:
Ok I'm also starting down the rabbit hole.  I downloaded USART HMI 1.65 and currently just using an [image language convertor ](https://translate.yandex.com/en/ocr)to read the fields and work my way around the software but I'm still in the prelim stages here but I'm starting to get a bit optimistic.   Refer to last link which will allow this to make more sense but the 4.3" resistive matrix screen coming w/ the Kobra now seems to be PN TJC4827X243_011_P07 and what do I see here when creating a new config:
[image]

Here is some more good info.  In Dec. some were saying that even if you were able to hack their util (USART HMI) that you would have some issues actually flashing the tft.  Details are [here ](https://unofficialnextion.com/t/tjc-editor-1-60-2-in-english/569/37	  With this said there is a cracked version that you'll find in a previous link (I think 0.86?) that is actually in English and functions.

However, there seem to have been [advancements]([image])....

This is the latest screen coming w/ the Kobra printers it seems.  It's also what AC will send you as a replacement
[image]

wabbitguy, 2023-07-30:
@TTVert thanks for the offer of the screen but at this point I'm going to pass. I'll stay in Marlin where I am at least familiar with.

I did try the download link for the exe file and I get a message "This XML file does not appear to have any style information associated with it." And that's all she wrote..LOL So not going to lose any sleep over it.

I suspect the editors for the DWIN or the TFT would be very similar. With the DWIN set you are literally using a bitmap graphic and drawing squares on for where you want things to happen. That gives you a coordinate that then has to be placed into the Marlin firmware so the DGUS routines they wrote know where on the screen to write, read from/to. Then it compiles a "configuration" file that contains all that info, plus a big file that contains all the compressed bitmap graphics. So yeah, the HTML editor from hell..LOL

Took me a while to wrap my head around the original DWIN editor, I'll leave the new screen up guys like you that have a better handle on it.

I found some firmware the Mainboard on the Kobra 2, it's compiled but Anycubic supplied a list of changes they have in it (V3.0.2). It's more advanced than what the Kobra, Kobra Plus and Max are. But there's no source code out for it either, just a compiled version. Sigh...:-)

Any way, if you need any help at all with Marlin, be happy to contribute any way I can.

And thanks again so much for the enlightenment. Before all your info and links I thought I was losing my mind (wouldn't be the first time but still...)...

Mel

TTVert, 2023-07-31:
> @TTVert thanks for the offer of the screen but at this point I'm going to pass. I'll stay in Marlin where I am at least familiar with.
>
> I did try the download link for the exe file and I get a [...]

Odd, all links worked for me a sec ago when I double checked.  Yeah I also thought i was  losing my mind when the DWIN FW update wouldn't flash.
Dave

TTVert, 2023-07-31:
@wabbitguy is there any chance your editor has the ability to save in an HMI format?  if so I may be able to get it converted to a tft for the JTC editor.
Dave

wabbitguy, 2023-07-31:
@TTVert the link may have something to do with my LAN or outgoing fibre network I use. I've had some things before that would refuse to work via links, like all those Google sponsored links when I do a Google search.

I've looked all through the DGUS editor I use and there's nowhere in it that I can see that does anything other than the DWIN stuff. It creates four files in the DWIN folder based on the stuff entered...they use the ICL format. No idea how that would transpose over to the TJC screen editor. Probably have to get the source from AnyCubic for the editor, thats what we had to do to get the DGUS setup. And they only gave out the screen stuff, not the LCD firmware that runs the show itself.

[image]

TTVert, 2023-07-31:
> @TTVert the link may have something to do with my LAN or outgoing fibre network I use. I've had some things before that would refuse to work via links, like all those Google sponsored links when I [...]

Yeah I have the same issue.  Use a pi-hole?  Those will do it.  Fair enough on the DWIN output.  I did a little research on converting an ICL into an HMI but not finding much.  AC just sent me another tft file for their TJC display.  Oddly it's older than the one I had sourced myself.  
[image]
Dave

wabbitguy, 2023-07-31:
@TTVert geez AnyCubic is right on the ball aren't they...LOL..

I forgot about the Pi-Hole cause yeah, I have one of those as well...so I disabled it and all I get. However I did find a link to a 1.64.1 version. No idea if that would work or not. Not even sure if the patches to English would work on it.

I have looked a fair bit at the conversion of ICL to the TJC requirements and found zip as well. It appears that which ever screen you have, that's the editor you jump in with and never the two shall meet. There does seem to be some links between TJC and Nextion (which Nextion goes to great lengths to deny) so it's like two farmers trying to sell the same cow. Sheesh.

 It's like AnyCubic using the KeilARM compiler for the Marlin source in favour of the more popular compilers. The KeilARM has weird syntax and it's caught me a number of times already, to the point where my forehead is getting sore from the face palms. LOL

And when I learned to write code, comments were considered smart coding. Because when I'd go back to a project six months or later it was easy to see where I was. Marlin is basically deaf and dumb. The DGUS library is equally matched. 

I'm pretty much convinced that if carpenters built houses the way the Asians write firmware, the first woodpecker that came along would have destroyed civilization....

[image]

wabbitguy, 2023-07-31:
@TTVert found these links. Who knows. Might work. Might not. Might be yet another bunny trail...sigh

[blog entry](https://unofficialnextion.com/t/tft-reverse-engineering-and-more/1152)
[Github](https://github.com/UNUF/nxt-doc)

Mel

TTVert, 2023-07-31:
> @TTVert found these links. Who knows. Might work. Might not. Might be yet another bunny trail...sigh
> 
> [blog entry](https://unofficialnextion.com/t/tft-reverse-engineering-and-more/1152) [Github](https://github.com/UNUF/nxt-doc)
> 
> Mel

Yep, they do know how to do it but they're keeping it close to their chests as the owner of Nextion (Patrick) is following them and making changes whenever they published progress.   I'm sure I could reverse engineer it but I don't have the motivation/time quite frankly.

Dave

wabbitguy, 2023-07-31:
@TTVert I fully agree with you Dave. I've been down that road way too many times in the past when I've been doing 3rd party software for devices. Every time I'd release an update the company would "tinker" with something for their next run. Which was a pain because they never even released an app for their own devices. They just didn't want their end users to be able to program their devices easily. Ugh.

I swear it would be easier in the long run to rip out the motherboard and display and put in something we can easily code and control. Granted that's well beyond the capability of 99% of the end users, but I know if my Max's have issues in that area, gutted they will be.

Stay calm and be well Dave...:-)

Mel
---

TTVert, 2023-07-31:
> @TTVert I fully agree with you Dave. I've been down that road way too many times in the past when I've been doing 3rd party software for devices. Every time I'd release an update the company would [...]

You as well.  if I get a fire under my butt I will let you know.  I ordered a DWIN screen to do some testing so I may get bored one day and try to figure this out.

TTVert, 2023-08-11:
@wabbitguy so I obtained what is supposed to be a compatible DWIN screen w/ the Kobra Max.  I was hoping for some insight.  It did not have the pin header and the pinout is different (and didn't come w/ the JST connector but I had a ton) so i'm just using jumper wires for now to power it and the data lines.  What I am experiencing on screen (On all FW, not just yours) is the screen boots (Although I don't get that boot sound) and it sits at the any cubic screen after the animation.  They all do the same thing.   Do these need the data signals connected to proceed past the anycubic screen I'm stuck at?  I have TX2/RX2 and well as TX4/RX4 and i tried both but w/ same result.  I even reversed TX/RX just in case the silk screen was mislabeled but no luck.  Given the stock FW doesn't work for this screen either I may be SOL but perhaps you can offer some input as to why it'd just sit here

[image]

Stock (And your FW) flash fine, no errors:
[image]

Kobra max pinout
Pin 1 +5VDC
Pin 3   RX
Pin 4   TX
pin 6 ground
[image]

DWIN screen pinout:

Pin 1 +5VDC
Pin 2   ground
Pin 3   RX2
pin 4   TX2
pin 5   TX4
pin 6   RX4
[image]

Thanks much, as you can see I really want to get your screen FW working w/ your board FW to take advantage of the zstep resolution, etc.

Dave

TTVert, 2023-08-11:
To be clear I got this non AC supplied DWIN screen, NOT the TJC working.  If you want the writeup I'd be happy to help.  Many others will be in this situation soon and they may want to move back to a DWIN screen if TJC development fails..  The screen was less than $25 shipped from china!  Doing all of that was quite easy compared to what you do.  I know little to nothing about writing FW/SW but I like to toy around, reverse engineer and teach myself things.  That's how I got into everything I know.  I did download DGUS 7.641 a few days ago and located a sample hmi folder on the internet to try to figure it out.  So far it's all greek to me but I will play with it some to try to understand it a bit more.  I guess right now I don't fully understand how the button value is defined when pressed.  I see Key value but it makes no sense how it would correlate to gcode.  I'm sure if I play around enough I may be able to figure it out w/ trial and error.

I did get and open your hmi, i assume all the colored squares are touch boxes?/  Did you see what I mentioned about the white fonts looking weird?  I "think" it's bitmap 133 and it says the font ID is 0 so I wonder if that is the issue I'm running into w/ the way that font looks?  I tried to preview from first page and I get this error regarding font0.   Perhaps I need to upload a particular font for the DWIN software to render the screen preview properly?
[image]

Dave

TTVert, 2023-08-11:
Here is what I have.  I have a word doc of it as well as the original (full size) images if you want them.

This is VERY basic/easy but may not be for the faint of heart as the PCB needs to be slightly trimmed.  You are not trimming near any traces so it shouldn’t be an issue.  You do get close to the screen so be careful.  If you are attempting this, you likely have more skill than I do so at your discretion you can trim the plastic tabs on the housing instead of the board in all places except the bottom corners which do need to be rounded slightly unless you feel you have enough material to trim the plastic for clearance for screen to sit flush at the bottom of the housing.  I didn’t explore that option as trimming the PCB was easy enough and locked the screen in place.
What you need.
1.	Soldering iron
2.	Some plastic cutting device, pick your poison.
3.	Some fiberglass cutting device.  Dremel w/ cutoff, file, knife, same deal as #2, it’s soft material.
4.	Allen keys to take screen housing apart (Come w/ the printer).
5.	A little patience, VERY little if I can manage!
6.	A small screwdriver, xacto, etc. to depress the metal pin to de-pin the ground pin on the female JST XH connector.
What to buy?
1.	Purchase the RESISTIVE screen PN: DMG48270C043_04WTR.  I got mine from [aliexpress](https://www.aliexpress.us/item/3256803104202500.html?fbclid=IwAR3Hxza8p_aFFHeVTxs4OrE8gJcK88c1kcQb7V7P23cSeinsHWqD4Ky0yaA&gatewayAdapt=glo2usa4itemAdapt) for under $25 shipped but feel free to shop around.
2.	JST-XH 6 pin male connector.  I had a ton but you can get a kit from [amazon](https://www.amazon.com/Connector-Housing-Adapter-Terminal-Connectors/dp/B0BZDCGJ32/ref=sr_1_3?crid=9AKXS0EUAJSW&keywords=JST+Sh+6+pin+male&qid=1691779708&refinements=p_85%3A2470955011&rnid=2470954011&rps=1&s=electronics&sprefix=jst+sh+6+pin+male%2Celectronics%2C105&sr=1-3) for less than $7
What to do? (Physical modifications to be made)
1.	You must solder a 6 pin JST XH connector onto the board to utilize the existing harness.  The board has existing holes for easy soldering of the connector, drop it in and go to town.  You can buy an entire kit cheaper than just one of these connectors with more than you’ll likely ever need for $6.99 from [amazon](https://www.amazon.com/Connector-Housing-Adapter-Terminal-Connectors/dp/B0BZDCGJ32/ref=sr_1_3?crid=9AKXS0EUAJSW&keywords=JST+Sh+6+pin+male&qid=1691779708&refinements=p_85%3A2470955011&rnid=2470954011&rps=1&s=electronics&sprefix=jst+sh+6+pin+male%2Celectronics%2C105&sr=1-3). 
[image]2.	
[image]
Bridge the 2 pins shown to enable the TTL module 
[image]
3.	Trim the PCB in two corners marked as well as notches to clear the plastic case.  Note you could also trim the plastic case in these areas instead, but I liked trimming the PCB as it locked the screen in place horizontally.  I did not use any screws during reassembly as the notches in the pcb combined with the two halves of the case which sandwiched the screen secured it very well.  If you do this right you will also have perfect alignment of SD card.
[image]
4.	Trim case slightly for screen clearance as you cannot trim PCB far enough in these areas due to mounting location of screen housing. 
[image]

5.	Clearance the back of the case for a slightly different location of the JST XH plug.  Just put the case half over it and you’ll see what you have to cut.   Use a file, Dremel, knife, etc. to make your cut.
  
[image]
[image]

6.	After you are happy with the fitment you now need to move onto depinning and moving the ground in the female JST connector.  The pins are just secured by a small metal tab.  Push the wire in a hair, depress the metal tab (with a knife, SMALL screwdriver, etc) and pull it out.  If you have issues use an exacto and gently slide it between the plug and the metal pin which will accomplish the same thing.
 
[image]

7.	Pinout.  **Note, you only be moving the ground pin from Pin 6 to pin 2.  Working left to right in this image.  
 
[image]

	TJC						DWIN
Pin 1 +5V (VCC)					Pin 1 +5V (VCC)
Pin 3 (RX)					Pin 3 (RX)
PIN 4 (TX)					Pin 4 (TX)
PIN 6 0V (Ground)				Pin 2 0V (Ground)
8.	 Flash wabbitguy’s awesome screen FW.   **Note, use an 8GB card as my 32GB card booted the update but it flashed instantly and nothing happened.  No indication that it wasn’t reading the data aside from the speed at which is flashed.  As soon as I put an 8GB card (Formatted FAT32) it flashed slowly and properly. 
9.	If you did it right you will be greeted w/ this and the screen will be perfectly centered in the housing.  You have a 3D Printer and likely some skill if you’ve gotten this far AND are putting your own custom FW on your printer, you could always cut out all the plastic in there and design and 3D print your own internal frame to retain the screen perfectly center if you don’t want to trim anything like I did.  The world is your oyster!
[image]

wabbitguy, 2023-08-11:
@TTVert I ordered one of the LCD's resistive touch directly from DWIN. Font 0, that can be built when you start up the DGUS editor and I have no idea which font it picks and why, but it might be just for viewing within the editor itself and not for the display. Again, I have no idea. Not like there's a manual for the DGUS editor...LOL

The font it generates is stored in the folder of all the other stuff (like the HMI file) but again, that might be just for viewing within the editor itself. And the error you got of the cells overlapping could be because of the font generated. The LCD display looks just fine to me.

I opened the HMI in the Dgus editor, clicked on the Welcome TAB after selecting screen #133, I generated the font 0, supposedly, no errors. When to the "Display" tab at the top and "Preview from Current Page" and it still whines that I should make sure font 0 is there. Huh. So colour me confused...sheesh.

That's a pretty damn good write up you have there. The icing on the cake is that last photo of success! You want to make a PDF of that, I'll put it on the hub here and you can refer or link to it, unless you want to put it on your own server in which case I'll still link to it for those who want to do it.

My LCD should be here by the end of August, or Halloween if it's shipped in a container...:-)

You know maybe you missed your calling as a programmer. You sure didn't miss it as an electronics buff!

Mel

TTVert, 2023-08-12:
You could just print the doc to pdf but I did it for you and put it in the same folder i shared previously. Feel free to do with the doc as you will, consider it yours.  I do some minor scripting for jobs to expedite client work but its VERY minor.  

I did create the font to get rid of the 0 font error but still got the "button configuration is intersecting" error.  How are you viewing your final product?  I thought it was under display | preview from first page but I only get this and cannot advance:
[image]

So you are saying your fonts don't look like this on the Settings | About screen?  Any letter which is wider is cut off.  (Capital D, W. etc.  Also some lower case spacing is odd..  Curious if this is somehow an artifact of this being a different screen.

[image]

wabbitguy, 2023-08-13:
@TTVert yup, I can handle the PDF. I don't normally use the Display-->Preview at all because it doesn't show anything other than a few areas:

[image]

The built in font is no doubt what the display is using to display the words. Font 0 in this case. What that comes set at from the factory I have no idea but there has to be a way to change it.

The English language screens start at 120.bmp and go up from there. The Chinese language ones are numbered from 0 to 119. If I select the 120.bmp this is what I see.

[image]

We'll have to look around and see if it's possible to change the font, or even how that's done. If it's in the actual firmware of the display itself, that's going to be difficult if not impossible because I'd imagine that firmware is closed source.

Mel

wabbitguy, 2023-08-13:
@TTVert ah ha... I found this link that describes how to build a font, which we already figured out, BUT the font that's created needs to be put in the DWIN_SET folder with the other files when you flash the display...

The default font seems to be the Microsoft Sans Serif 16 point and changing it to another point size doesn't really seem to work, but maybe this'll help.

https://how2electronics.com/smart-home-controller-using-dwin-hmi-display-esp32/

TTVert, 2023-08-14:
> @TTVert ah ha... I found this link that describes how to build a font, which we already figured out, BUT the font that's created needs to be put in the DWIN_SET folder with the other files when you [...]

Yeah i saw a video on font scaling last night  I just had a moment now so I created a new font and scaled it.  Looks better but not perfect, will find a new font and then generate the HZK file.  Then it just needs to be included in the DWIN_SET folder.  For anyone who has your existing setup and cares can just put that HZK file in a DWIN_Set folder (with nothing else) and it'll push that file to the screen and the font is changed.  This is where I'm at now with a scaled Sans serif (Default font).  It's better but not perfect.  I made some changes but they didn't seem to make it any better.  Not sure why the spacing for certain letters if off, it's something I'll have to play with later after i understand this program a bit more.
[image]

Dave

wabbitguy, 2023-08-14:
@TTVert found this, might help and its got an original font file as well (or so claimed)...

https://github.com/Quint097/DWIN_Font_Tools/tree/main/HZK%20Font%20Files

For the generation itself, I found this description...

We can close the window and open the 0# word bank generating link in the DGUS config tool area. The font editor is used to create font sets which are later used to display characters on your DWIN display. You can create multiple font files and you can refer them to different controls on your DWIN display. Select your preferred font, then select a wide character, for example, G, M or W. Then adjust the scale and shift values until all letters fit their frames properly. Then press Create 0_DWIN_ASC.HZK. When the process is finished, open your file browser and go to the folder where your DGUS executable file is located. You will find the freshly created font file here. Copy it over to your project’s DWIN_SET folder.

The text boxes for the displays (as designed) is 14 pixels high. So that's what you got to work with. I was looking for a mono-spaced font (like standard ASCII font) but didn't see anything. But changing the values did make all the characters at least fit.

Hope we get this sorted out...

[image]

TTVert, 2023-09-01:
> @TTVert I got my LCD display today, now to find time to get around to testing it... Prior I did rip apart the display on my Max to pin it out and see what I could find for markings. As it turns out, the displays were made in 2021. They supported WiFi at that time as well so they aren't the same as the one I ordered.
> 
> The ones AnyCubic used are different so I managed to track down the ones they did use. In India of all places. I don't know if they are legit DWIN displays or knock off copies.
> 
> [LCD- DMG48270C043_03WTR](https://robu.in/product/dwin-hmi-4-3-inch-tn-lcd-resistive-touch-display-copy/)
> 
> On Aliexpress there are scads of _04WTR displays, but only one 03WTR I found with rather silly shipping costs.
> 
> [Aliexpress - 03WTR](https://www.aliexpress.com/item/33037525947.html)
> 
> I downloaded the TL5 documentation and "font 0" is supposed to be installed in them as a factory default. Still not sure about that.
> 
> Any way, just thought I'd update what I found out.
> 
> Mel

Awesome!  I've been out of town for a while but I'm curious how Font 0 can be preinstalled as the font pack has to be built as it's scaled to your defined size isn't it?  BTW, I think i paid more for shipping than the item also :D  I wonder what fun could be had w/ that wireless interface?

Dave

Dave

wabbitguy, 2023-09-01:
@TTVert when I got the display I started to pin it out, and I ripped mine apart to see what was what. Thats when I found the differences. Before that I was just assuming all these displays were the same. Turns out, nope. I then started to look for boards made in 2021 and what the physical layout was and thats when I found the ones I did.

I remember that it was in the firmware where I saw the mention of Wifi. Its in dgus_tft.h:

#define PAGE_WIFI          (12+PAGE_OFFSET)

So it'd be page 120 + 12 = 132. If you check the BMP for 132, you'll find it blank. And the Kobra 2 Max now includes wifi.

According to the sellers, font#0 is always installed and with the TL5 documentation it says the same thing. They don't give much if any info on what font#0 is based on, or anything much about it. They do tell how to generate your own fonts in the TL5 documents but thats about it. And it's a long boring read...ugh...

The DWIN they use was a 03 because it supported the Wifi board. Which looking at the connections might be a ESP8266-01 or something akin to that. Or they have an ESP8266-12 mounted on a breakout board like these guys did:

[ESP8266 Breakout board](https://climbers.net/sbc/esp12-test-jig-breakout-board/)

Just instead of the pins, there'd be solder pads. So a custom board layout.

The other area thats unpopulated on the original display might be for the capacitance display instead of the resistance one. On the new display, they have an I2C bunch of pins there for something. But we'd have to find a breakout of the PCB to figure it all out.

Mel