wabbitguy/Kobra_Max #22: New motherboard revision - breaking changes?

3ditguy, 2023-07-29:
I received a new motherboard today to replace one that failed under warranty. The board model is Trigorilla_Pro_B_V1.02 and is easily identifiable by its USB C connector. It has firmware v 3.0.8 and there is already one report of someone flashing old firmware and running into issues. Worth putting a big warning in the readme? 

https://www.reddit.com/r/anycubic/comments/14z145y/kobra_max_usb_c_not_recognized/

[image]

wabbitguy, 2023-08-02:
@3ditguy I've only found all this new motherboard and LCD display stuff out within the last two weeks and there's been zero information about it from Anycubic (no surprise there). I've been trying to figure out a way to differentiate the boards for the end users but your photo fixed that in a heartbeat.

The LCD is no longer a Nextion unit, it's a JCT panel, and there's no screens or setup for it like was provided for the original display. There's no way to tell the displays apart except to remove the screen and read the back of the display motherboard.

I've seen a number of people with newer Kobra Max's and the new motherboards (but still original displays) and AnyCubic has provided them with firmware for their new/replacement motherboards. The source code of which doesn't exist on the Anycubic Github site, and I honestly don't know if it ever will. Hard to keep ahead of production run changes.

Anyway, you have my sincerest thanks for the photo, you just made my life so much easier. I'll update the readme with it.

Mel

closed 2023-08-02 by wabbitguy

3ditguy, 2023-08-03:
@wabbitguy 
Another interesting bit of info. Since getting the new motherboard (which did resolve the original issue) I have been plagued with small layer shifts in X (very frequent, roughly every 15 seconds). They appear to be heat related with issues showing up reliably after around 2 mins of printing and persisting if you cancel and immediately restart a print but temporarily going away if you allow the printer 30 mins or so to cool before starting a new print.

On raising this to anycubic via the seller:
"If the USB port of the motherboard received is -C port, this is a new motherboard, which needs to be used with the new motor"
They are sending me:
"The package is sent separately, the X motor and 2 motors are in transit"

Seems like this could be a painful transition.

wabbitguy, 2023-08-03:
@3ditguy okay that's just plain weird. I'm well versed with overheating stepper drivers and that's one of the problems with a Y Axis on the original Kobra Max if the jerk/accel is set too high. I'd hear a loud "clunk" (sounded like someone tossing a baseball in an empty garbage bucket) and the Y shift would occur. I did a lot of tracking with currents for the stepper drivers and making adjustments to the vREF for the Y axis, but on one of the Max's anything over 70mm/s in speed and it would happen, usually about an hour into the print. Stepper driver heat signature was well within range.

My thought finally boiled down to I wondered if these were cloned TMC2009's and not the higher quality ones. With no way to tell, I removed the Y axis chip (and I don't advise anyone to try it; I have the tools to do it but it's still difficult) and replaced with one I pulled off a Watterott board. Powered up the printer and it ran at 80mm/s for hours, no problem at all. Which doesn't mean diddly squat because I still don't know if it was a clone chip or just a substandard one.

The motherboard cooling fan could use a swift kick in the butt, I've been tempted to dump in a 92mm Noctua fan that moves some air and is quiet. Plus add the larger heatsinks to those stepper chips. But I've done neither at this point. Just on a list of things to do when I get bored...LOL

The 2209's should be able to power any 1.68A motor there is. So...now I'm wondering if the new motherboards can actually be programmed for VREF like the original or if they stuck a resistor on the line to set it to a value and the firmware can't control the drivers. If that were the case then yeah changing the motors would reduce the current draw. But without the schematic/source code, it's all a duck shoot. Meh.

I really hope you get that thing tamed and get back to printing. Thanks for the update!

Mel