wabbitguy/Kobra_Max #23: Bed crash after premature termination of print

zebrabunny, 2023-08-21:
Bed crashes after early termination of print (from SD card slot)

I start a print, realize something is wrong, hit stop print, then confirm. when the machine stops the bed moves forward rather than to the home position and crashes noisily. should home instead? or even better, do nothing.

wabbitguy, 2023-08-30:
@zebrabunny at the many requests by end users, the bed moves forward so you can rip the failed print off the bed. It made no sense to them to move to the "home" position because then they had to grab a hot bed and drag it to the front to remove the failed print.

As for why it's "crashing", my guess, since in all our testing that hasn't happened, is that the printer for whatever reason has no idea of what it's current Y position is. Which doesn't even make any sense because the instant the GCODE starts and the print homes, it knows where it is.

zebrabunny, 2023-08-30:
Cool, I will take it as offered :) can you answer me a little question, in
what file is your filament load-unload code? I've found two variations
neither of which are germain to the Kobra Max :) I have done a direct drive
mod so I want to shorten the load length
Thank you for the nice firmware :)

wabbitguy, 2023-08-30:
@zebrabunny it's in two places actually, just depends on which load-unload code you want to change. Or both.

If you're talking about the LCD load, unload filament buttons, the GCODE for them in is dgus_tft_defs.h and look for:

#define MEL_cmnd_auto_unload_filament
#define MEL_cmnd_auto_load_Filament

Those are the GCODE's for loading and unloading using the Kobra Max LCD menu.

Now if you're talking about a filament change (M600) that's a completely different kettle of fish. For that you need to look in Configuration_adv.h and look for:

#define ADVANCED_PAUSE_FEATURE

In there you will find ALL of the measurements for the distances used for a filament change, plus the speeds and accelerations.

Once you change all of those items, then it's a recompile and test to make sure it works. There's no way to change all of those items without a complete recompile, i.e. it's not a simple GCODE change (although you might be able to change the GCODE lengths with M code for the load/unload, you have no control over the purge, speeds or acceleration and a direct drive is a non-standard puppy on these things).

If you get stuck, you know where I am.

Mel

zebrabunny, 2023-08-30:
have already forked off a few things, the compiler is purring. thanks for
the info

zebrabunny, 2023-08-30:
(I was talking about the MEL stuff, got to the rest already)

wabbitguy, 2023-08-30:
@zebrabunny good to hear you can do the compiling. I tend to end up do a fair bit of custom compiles for others. Not that it's that much of an issue for some of the changes but it all takes time...

If you want to update me with the M600/Gcode changes for a direct drive I'd appreciate that. It save some others who have converted to direct drive from having to measure all the stuff out. And of course the direct drive you changed to would help as well.

I'm going to close the issue, but you can reply here any time even after it's closed and I'll get a notification of it.

Mel
---

closed 2023-08-30 by wabbitguy