wabbitguy/Kobra_Max #24: New hothead

rahafuf, 2023-10-02:
is there any way you can change the thermo for the hothead to Thermo #30 for 300 degrees? been trying to do it for a few days now and can't figure it out. The thermo is NTC100K B3950. It can work up to 300℃ for a long time.

wabbitguy, 2023-10-04:
@rahafuf originally Anycubic used a 5 x 5 grid, the newer firmware uses a 7 x 7 grid. What a lot of us have found is that the bowden tube will pull on the hot end/strain gauge as the levelling takes place and skew the accuracy or cause a fail during levelling.

Once the new firmware is installed, it always requires a factory reset. That's outlined in steps with PRONTERFACE via USB in the readme for the Max here.

If the firmware was installed according to the steps but it fails, it'll usually end up to be the bowden tube hauling on the hot end, a faulty strain gauge, or a wiring issue between the hot end and the motherboard. Be nice if Anycubic used wire that was actually designed to flex instead of the garbage they use now.

Re the thermistor, changing the type is a simple job for me. But it would have to be a custom one up firmware because it would be unused by any one else. However, before that can be done, you'd need a hot end that was actually safe to use at 300C. The stock hot ends are going to off gas like crazy at 250C and higher. To the point if you have a pet bird it's going to have a lifespan that can be measured with an egg timer. PTFE tubing isn't any good for temps over 250C, regardless of who makes it. It melts, of gases and causes problems with health.

Thus, I'd need to be reassured that a proper hot end is in use before I make any unsafe changes to the firmware.

closed 2023-10-04 by wabbitguy

wabbitguy, 2023-10-18:
@rahafuf Looks like the standard hot ends for the Kobra Max, I've boughten several of them (rebuilt them all because of the clogging issues they have with the stock set up. Simple fix really. End of clogs.

That hot end only has the thermistor rated for 300C, it's still the old fashioned PTFE tubing shoved all the way to a hopefully tight fit to the nozzle. So it's going to off gas like crazy at anything over 255C. You need a bi-metal heat break if you want temps higher than 250C with that setup. The bi-metal probably looks like the one in the photo (you just have to check the measurements against the stock heat break to ensure diameters and overall length is the same.

[Bi-Metal heatbreak](https://www.amazon.com/Yunbotong-Bi-Metal-Heatbreak-Titanium-Temperature/dp/B0991QJ3L6/ref=sr_1_2_sspa?crid=1MEYFA8NHTZE5&keywords=bi-metal%2Bheat%2Bbreak%2Bvolcano&qid=1697666088&sprefix=bi-metal%2Bheat%2Bbreak%2Bvolcano%2Caps%2C118&sr=8-2-spons&sp_csd=d2lkZ2V0TmFtZT1zcF9hdGY&th=1)

As for the firmware what do you need modified in the firmware?

rahafuf, 2023-10-19:
only thing would be the temp change. everything else is great. is that the bi-metal heartbreak I need? if so I will order it today.

rahafuf, 2023-10-19:
probably going to order these but want to make sure it works.

[BZ 3D I3 Mega Heatbreak Vyper](https://a.co/d/jiDkbtv)
[Capricorn PTFE Bowden](https://a.co/d/6lRChue)

wabbitguy, 2023-10-19:
@rahafuf the BZ_3D_i3_Mega_Heatbreak_Vyper to linked to is the correct size (I have some of them and measured them against the stock heat break and they are the same). So they will work in the hot end.

The Capricorn XS tubing you linked to is a poor choice. Problem with it is the ID is 1.9 and if the tubing has any bend in at all that bend will reduce the ID and grip the filament a little too tight. If you want to use Capricorn tubing look for the TL version of it. It works much better with Bowden systems. Although, you don't need high temp stuff anyway with an all metal hot end. The TL tubing is more transparent than the XS stuff.

It'll be interesting to know if the thermistor will survive 300C temps since 99% of them are rated for a max of 270C. I've seen more than a few wires lose their insulation coatings at the 280C range. Even the genuine E3D V6 Volcano is only rated to 295C unless there is a thermocouple upgrade. Hard to imagine anything used by 3D printer companies as being related to quality. But, well, if it goes up in smoke, least you'll know.

For the firmware, I assume you want as close to stock as possible with the temperature increased to the 300C max.

rahafuf, 2023-10-19:
ill get the heatbreaks ordered. What tubing would you recommend? 295 would be great

wabbitguy, 2023-10-19:
@rahafuf for the tubing (PTFE) you could basically use anything. Doesn't matter if it's the budget stuff that comes with the printer or the higher end Capricorn TL tubing. It never heats up, never gets near the hot end so whatever you have will work fine.

295 would be good, but it's going to be a test as well because companies don't use high end thermocouples in these printers. A lot of them won't even measure correctly after the 270C range and the tables in Marlin the thermistors use is going to be shy of a lot of data up in that temp range. So...if it holds together it will print (but who knows for how long) otherwise there'll be a bit of smoke and it'll be game over.

I changed the thermistor to the NTC100K B3950 and bumped the temp to 300C. Other than that it's the same as the 1.8.4 release version, albeit with a changed "Printer" name.

[firmware.bin.zip](https://github.com/wabbitguy/Kobra_Max/files/13047128/firmware.bin.zip)

rahafuf, 2023-10-19:
awesome. ill test it now and report back

rahafuf, 2023-10-19:
getting a hotend NTC abnormal error

rahafuf, 2023-10-19:
installed but the E temp only goes to 260 instead of the 295

rahafuf, 2023-10-19:
do i need to do anything once it installs the firmware?

rahafuf, 2023-10-19:
ended up purchasing this
[Upgrade V5 J-Head Hotend](https://a.co/d/hXBzt7H)

wabbitguy, 2023-10-19:
@rahafuf The firmware has the max set to 300C, the LCD controls could be as brain dead as ever though so if you're trying to set the temp to 295 with the LCD, it may not work. But if you connected up PRONTERFACE via USB and set the temp to 295, you'd find out pretty quick if the firmware or the LCD is the brain dead part.

That hot end all metal you linked to, well, it would really be nice if they mentioned what thermistor they were using instead of some stupid generic NTC 100K. Which is typical for the silicon bed heaters but not for thermistor cartridges.  I loathe and detest these sellers. Ugh.

rahafuf, 2023-10-19:
ok so it looks like using prnterface i can in fact go to 280*

ill ask on it about what thermo it is. hopefully its not a generic

rahafuf, 2023-10-19:
@wabbitguy so i set it to 280 degress and I'm getting a "Hotend NTC abnormal, please check it and wiring!" error

rahafuf, 2023-10-19:
[image]

wabbitguy, 2023-10-19:
@rahafuf can't find anywhere in the LCD code where it's doing any limiting on the hot end to 265C. It uses the define I set at 300C as the max. If you use the filament in/out in the LCD those are completely different and limited in a completely different way (they are GCODE's not defines). Same with the "PreHeat" they are also completely separate and have no bearing on the Max temp define.

So slice up a model of something and feed it some gcode to see if it goes to the 295. If you need the bed hot as well, it'll take about a week to heat up to temp's of 100C...LOL

rahafuf, 2023-10-19:
i didnt touch the screen at all. set the temp with pront and that popped up and showed on the screen

wabbitguy, 2023-10-19:
@rahafuf okay, MaxTemp triggered. mmm... wonder where in the world they buried that in Marlin...

rahafuf, 2023-10-19:
think i need to do a PID for it?

rahafuf, 2023-10-19:
i found this
`  if(millis() < 5000) {
    return ;
  }

  _temp_error(heater_id, PSTR(STR_T_MAXTEMP), GET_TEXT(MSG_ERR_MAXTEMP));
}
`
its in the temperature.cpp lines 856-861

wabbitguy, 2023-10-19:
@rahafuf yeah from what I found, with much digging, the thermistor you define sets the maximum temperature you're allowed to use. So that's what I'm trying to track down at the moment. It's buried in some include file someplace. Obviously it's not meant to be toyed with...LOL

rahafuf, 2023-10-19:
i might need to do a PID tuning since the firmware change. 

is this the correct one to do?
M303 E-1 C8 S60

wabbitguy, 2023-10-19:
@rahafuf you can try this one. It might go to 299C (thats its defined max). Hard to say. 

For PID for a hot end:

M303 E0 C10 S###

Where ### is the temp you want it most stable at. Course this is going to probably lock up the Kobra Max's LCD since it's not designed to handle any PID stuff. So remember the values you get from Pronterface, cycle the power on the power and then you'll have to enter in the values with pronterface again via codes.

[firmware.zip](https://github.com/wabbitguy/Kobra_Max/files/13048170/firmware.zip)

> i might need to do a PID tuning since the firmware change.
> 
> is this the correct one to do? M303 E-1 C8 S60

rahafuf, 2023-10-19:
what do you mean enter the vaules on pronterface?

also what temp should I do?

rahafuf, 2023-10-19:
reached 270 fine
reached 280 fine
reached 290 fine
reached 295 fine but fluctuates 2-3 degrees around it but I think 275 is the most ill do.

wabbitguy, 2023-10-19:
@rahafuf let me just hazard a guess that you're new to 3D printing and all the tweaking and tuning...LOL..we all had to learn stuff at some point, I started in 2015 so all this is nothing new and I've been working in Marlin since then. Mostly on deltas cause they're way easier for me than these off shore cartesian boxes.

For the PID tuning, the idea being is that you set the temperature as the norm for what you print at. So if you plan on doing nothing but printing at 275C then do the PID at 275. Usually the range is fairly accurate if you go +-30C. So if you tune for 245C, you should be accurate from 225 to 275.

That would be: M303 E0 C10 S245

As for Pronterface, you have to do PID tuning from its terminal panel via GCODE. There's no way to do it any other way. And it will lock up the printer when it's done. So you need the 3 values it gives you, write them down. Then power cycle the printer, let Pronterface connect again, then from the same terminal panel you have to enter in the 3 values again with GCODE commands and then an M500 GCODE to save them.

The link shows the steps used.

[PID tuning](https://printermods.co.uk/blogs/guides/pid-tuning)

rahafuf, 2023-10-19:
somewhat new. new to modding them atleast. I've used prusa before but never had to do a PID tuning or anything like that.

wabbitguy, 2023-10-20:
@rahafuf I have a PRUSA as well (MK3S/MMU2S) thing is a toaster. I plug in filament, it prints. A work horse by all rights.

My deltas (all scratch built from a bunch of parts; never owned a commercial one) are work horses as well.

I will say I'd be hard pressed to ever invest in an Anycubic printer again. Mine works fairly reliably but with the stock 0.4 nozzle and the huge size of the bed it's pretty slow for large stuff.

rahafuf, 2023-10-20:
honestly i only got it for the size lol. wish there was others at this size for around the same price

wabbitguy, 2023-10-20:
@rahafuf I got mine for a bit of a deal and it had a large bed because I was making some outdoor projects and needed a large size. When I got it I realized Anycubic doesn't want to put any standard features in the firmware because that costs them money in tech support. So I nagged them until they released the source code.

My criteria from now on is if there is no source code for the printer, it won't even be on my list. At least that way I can attempt to fix some of the problems.

rahafuf, 2023-10-20:
im glad they atleast did that for the people who have it. wish they would release info on the new TJC screens. they are way to bright xD

wabbitguy, 2023-10-20:
@rahafuf I know what you mean. The TJC editor is significantly different than the stock display.

They haven't released the firmware for the newer Tri-gorilla ProB_V1.0.2 boards (I've worked with a few people who have those boards and it's a mess) so add that to the TJC displays too.

I mean what self respecting company changes the motherboard and display for no good reason. And the 1.0.2 board is a sub-standard board as well. There's no UART for comms to the stepper drivers, all the drivers are 2208's, and the list goes on.

In the end it's about profit, not the end user experience. I've asked them numerous times for their 1.0.2 firmware but they ignore me. The only thing they respond to is money. I think they like publicity, even bad publicity.

Any way, good luck with your project!

rahafuf, 2023-10-20:
thank you very much for your help brother! is there any way you can show me what you changed? I'm wondering if I had it correctly before when I did it but I only tried to change it on the printer itself and not through an interface like pronterface

wabbitguy, 2023-10-20:
@rahafuf  two changes...

Configuration.h

#define TEMP_SENSOR_0 11
#define HEATER_0_MAXTEMP 314

If you hunt in the source code, look in Marlin/src/module/thermistor and you'll find the tables for the thermistors that are defined. If you open one of the files you'll find a number in the arrays that defines the maximum temp the thermistor will support. Don't expect any comments in the files, Marlin programmers apparently lack the ability to comment their work very well...sadly...

For example if you look at thermistor_11 you'll find:

{ OV(  31), 314 }

That 314 is the MAX temp, sorta. Marlin automatically subtracts 15C from whatever the max temp is in those thermistor files. So take that 314 and subtract 15 from it you get 299C. Thats thermistor #11 max temperature you can set.

Take thermistor #5 and look at its file. Max temp is 285C (max is 300 in the table so 300 - 15 = 285).

You can open each of the thermistor files and you'll find temps from about 230 to the 299 range and thats it. These things were never designed to push up in the 250 - 300 range because of their flammable natures (300C is ignition for a lot of things in the real world; so a fan failure would be not a good thing), and the fact these are all basically low budget printers with iffy components to start with, not to mention the hot ends have teflon tubing shoved up against the nozzles that off gasses like crazy at anything over 260C. I mean who wants a 3D printer that needs a respirator...

For the PID tuning, the only way to do it is through pronterface. The LCD has zero support for PID tuning, it's not in the LCD code at all. You either change it in firmware (which then gets saved to EPROM from a fresh install like when the printer was new and you were doing the levelling and so on) or you adjust it from PRONTERFACE with GCODE and tell Marlin to save it to EPROM.

wabbitguy, 2023-10-20:
@rahafuf oh and if you do an About information for your printer to see the firmware version, you will see a unique printer name as well.

rahafuf, 2023-10-20:
awesome. thank you brother. Got the full metal hotend installed today. only lists the NTC100k as a thermo. i shouldn't need to change anything in the firmware right?

wabbitguy, 2023-10-21:
@rahafuf hopefully you won't have to change anything. NTC100K is pretty generic.

rahafuf, 2023-10-21:
ok cool. thanks so much. trying to figure out this startup tone thing now. says to use https://www.apronus.com/music/flashpiano.htm but  cant figure out how to do it lol

wabbitguy, 2023-10-21:
@rahafuf it's basically a MIDI keyboard, and the dgus_tunes file is the note and duration of the note in an array.

They have a defined list of the notes in the dgus_tunes that correspond to the notes on that keyboard. So you need notes, timing signatures 4/4, 2/4 whatever time signature you're playing in for the note duration and the pauses (rests).

It would help to be a musician to be able to play and transpose those notes you play into the data for the "music" (dare I call that start up chime music? LOL)...

All pretty simple stuff if you're a musician and a programmer (I've been a guitarist for the last 60+ years).