wabbitguy/Kobra_Max #29: G29 not accepted

donbarthel, 2024-08-15:
With the stock firmware I can level the bed with the sequence G28 G29 (Home All then perform a mesh probing). With this custom firmware it freezes before completing the G29 at a random position.

The reason I want to level by issuing g-code commands from a command line is so that I can perform the leveling at whatever nozzle and bed temperature I choose. I found that with the stock firmware the leveling initiated from the panel would always cool to PLA temperatures (and now this is part of my routine).

closed 2024-08-19 by wabbitguy

donbarthel, 2024-08-19:
> @donbarthel <https://github.com/donbarthel> I've never made any changes
> whatsoever to G29 GCODE handling routines, they are identical to the last
> revision of the firmware AnyCubic did for the Tri-Gorilla PRO_A 1.0.4
> boards. Matter of fact I've never even looked at them til now, just to
> verify that they are the same. The configuration define sets the temp to
> 190 because AnyCubic defines the wrong thermistor used in the hot end (it's
> 10-20C higher than the reported temp).
>
> One major difference with this firmware over the stock firmware is the
> mesh matrix. Stock is 5 x 5, mine is 7 x 7. At an 8 x 8 matrix the
> Tri-Gorilla 1.0.4 motherboard runs out of memory and will randomly crash
> all over the place. Even a 7 x 7 matrix is getting close to the memory
> limit of the board.
>
> When updating the firmware, the EPROM has to be wiped out to remove the
> old firmware's settings. That has to be done with PRONTERFACE (Printrun)
> via USB. Not sure if you did that step or not.
>
> I never bothered to change the levelling temp because I levered it once
> ages ago, then adjusted all the individual points with M421 and a
> micrometer mounted on the hot end case to get it as flat as possible. I
> print ASA, HIPS, Nylon, PETG and some PLA (stock hot end with PTFE throat
> mod; 0.4 nozzle), bed from 60 to 105C. I have a double sided PEI textured
> flex place I put on the top of the glass so that's what I printed on.
>
> On my Max's the only time I've had either printing or levelling issues is
> with the cable from the hot end to the motherboard breaking. Then at some
> random point, the whole thing just stops. There's an error generated but
> the LCD display has no capacity to show them and it's closed source.
>
> If you want I can dumb down the mesh size to stock and you can give it a
> go. Beyond that, not sure what else I could do.

Really, I'd just like you to try a G28 G29 sequence to see if yours
completes successfully or not.

wabbitguy, 2024-08-20:
@donbarthel I would but I gave away the three working Kobra Max's that I had. I just never used them after I bought a QIDI. The only larger bed slingers still here are two ancient Chiron's that refuse to die and they see little use as it is. They probably need to go to a home to see some use as well one of these days.

I largely got fed up with Anycubic when they changed the motherboards in the second release of the Max printers and their tech people didn't even know about the change. And talking to their developers is simply counter productive in that they've never used a 3D printer themselves and their mandate is simply to make the printer "work".

Once I started developing the firmware I could see exactly why they hacked the daylights out of Marlin by omitting  huge sections of libraries/code sections. The motherboards simply don't have enough memory to hold the entire Marlin code. The ancient ATMEL MEGA 2560's 8 bit boards have more memory than the 32bit crud pushed out by Anycubic. sigh...

donbarthel, 2024-08-20:
Ok, I understand. Can you point me in the right direction, what tools do I
need to compile the firmware?

wabbitguy, 2024-08-20:
@donbarthel as I said, I could compile whatever you want changed but if you want to do it yourself, you need to use a specific compiler.

KeilARM, there's no other compiler that will build a binary. The source code itself is based off Marlin 2.0.8 and Marlin 2.0.8 bug fix at that (with some old Vyper code tossed in as well)... so there's a lot of stock Marlin stuff that's just omitted because the developers are less than stellar.

There's a plethora of steps to get KEIL to run, and you get a one year license for it when you register. The steps are on Reddit:

https://www.reddit.com/r/anycubic/comments/y2waxu/tutorial_how_to_build_anycubic_marlin_source_code/

This will allow you to build the firmware. It will NOT allow you to do ANYTHING with the LCD display. The LCD display is basically a web page monster where you define "pieces" on a page, and then in the Marlin source code refer to those "pieces" to actually make something show up on the display via the serial protocol it uses. Thus it also takes a different custom compiler to make the screen work as well.

Good luck!

donbarthel, 2024-09-04:
Ok, weird, your version of the firmware isn't crashing when I issue a G29
from the command line anymore. I've done it on two separate days without
issue. FYI. Thanks again for the improved firmware.

wabbitguy, 2024-09-04:
@donbarthel that is strange. Almost like something got reset in the EPROM or something and now it's happy. Any way, happy to hear the firmware is working out for you!

Mel