wabbitguy/Kobra_Max #31: Observation

david-taylor, 2024-09-24:
As I have a Standard Kobra, a lot of your changes are also relevant throughout the Anycubic fdm family. I have noticed on the later sources they have released on the different models, some of the changes have been implemented. I was wondering if it is possible to have something like a diff file separate for each mod. Such as the M600 pause for colour change, 7 instead of 5 point level setting, speed-ups, MAT-MOD whatever that was, etc. Adding commented documentation helps too. Oh and thank you for what you have done so far.

wabbitguy, 2024-10-09:
@david-taylor I have V1.8.4V for the standard Kobra available. I don't show it anywhere here on my Github site because I don't own the standard Kobra. However I have a friend who did and they beta tested for me so I only release it for those who ask for it. It's using a 5x5 grid for levelling since it's a much smaller bed size and really didn't benefit from the 7x7 the Max has.

If you want, I'll compile up a custom version for you. Just need to know the name of whatever you call your printer (i.e. like a nickname).

Mel

closed 2024-10-09 by wabbitguy

david-taylor, 2024-10-09:
Hi Mel,

That would be great. I have no special names for my devices (besides
bastard when they don't do as asked). Surprise me. I am trying to setup
Ubuntu 20.04 to do the compiles hence the reference to the diff files. I
guess I could just run a comparison. It appears the later printers have the
same M600 issue as they hi-jacked it for status save on power fail'.

All the best from an Aussie.
David

wabbitguy, 2024-10-09:
@david-taylor okay... AnyCubic hires contract programmers. Which is why you tend to see bug fixes for "x" months and then nothing. That's because the contract is up and it's too expensive to hire programmers to fix bugs when they were the ones that developed the code in the first place.

The Kobra code is ALL based of the original contract of the Vyper code. Which is based off a Marlin 2.0.8 "bug fix" branch, not even a release branch. The second thing AnyCubic pimps is the 32bit board, which is wonderful IF it actually had enough memory to support all the Marlin functions. But, it doesn't. The Arduino MEGA has more memory than the Tri-Gorilla boards do. And that's what you learn when you lift the hood and kick the tires...LOL

At any rate, here's the firmware for you. Standard Kobra 220*220*250... And don't blame me for the name, you did say surprise and after all I'm a  Canuck...

Update procedure will be exactly the same as a regular Max except you shouldn't have to wipe the EPROM via Pronterface/USB... (if I remember right)... You should also be able to use the new UI found on here as well so long as it's not the TJC display that Anycubic replaced the original one with.

[firmware.bin.zip](https://github.com/user-attachments/files/17302029/firmware.bin.zip)

david-taylor, 2024-10-12:
Hi Mel.

Just tried the image. Strange load that just locks at the Anycubic splash
screen. No further. Reboot. Same thing. Size is < 200k. Original image is
just short of 500k. I think it might be truncated. I was able to reload the
firmware so not bricked.

Cheers, David.

wabbitguy, 2024-10-12:
@david-taylor that sounds normal for an update. I found about 50% of the time the printer will sit at the startup screen forever during an update. The LCD is actually running from a serial port on the motherboard not the typical higher level protocols we see with some of the LCD's.

I recompiled and the firmware is 198K (199KB)... The original image from AnyCubic has a lot of zero's on the end of it if I remember correctly.

I know some users haven't reset the EPROM for the 7 x 7 mesh levelling on the Max and that causes some issues. I also know that if there is a USB cable in the printer with a host or computer connected it will foul up the updates as well. But the Kobra firmware I did only adds in the missing stuff.

Here's a fresh compile (toss the old one). Also make sure this is the only file on the SDcard and at the root of the card :

[firmware.zip](https://github.com/user-attachments/files/17348591/firmware.zip)