wabbitguy/Kobra_Max #34: Printer pauses for filament changes when shouldn't

leonet99, 2024-12-10:
I'm using Kobra_Max v1.8.4 firmware (thank you btw!)
I'm running some retraction tower tests set up in PrusaSlicer using 5 virtual extruders to print the 5 stacked model segments for the different retraction settings in one print (set up in printer extruders, disabled in filament overrides). All gcode is disabled in the slicer for filament changes and no color changes, tool changes, or pauses have been inserted in the slice preview window.
Kobra Max pauses the print every time it ends one segment before the next and takes out the filament as if it looks for a filament change. I put the filament back in and resume printing just fine, then the printer pauses again and the same way when it finishes the current segment before moving to the next.
I am also performing the same type of test with my smaller Kobra printer. I have actually worked in parallel to set up both printer tests in very similar fashion (their custom gcodes are also almost the same except for their obvious differences), but the Kobra printer keeps printing all the test model segments with no pause until the end.
I looked at the actual print gcodes on both printers and don't see anything to prompt that behavior, other than a tool change between each segment (T0, T1, T2...), but thta is teh same on both printers gcode, while only the Kobra-Max pauses and needs the filament reinserted and the print needing to be resumed...
Why do you think that happens and what should I do to avoid this and have the printer keep going until the end? 
I have a lot of tests going and babysitting for this issue is very time consuming ...
Thank you!

wabbitguy, 2024-12-12:
@leonet99 you've found what the firmware is supposed to do as I programmed it. In the firmware anytime it finds T0, T1, T2 etc it automatically kicks in a filament change. There's no way to over ride it.

Now, for the reason why it does that.

99.9% of the single nozzle/filament world of printers are only able to do "layer based" filament changes. Which is fine but it's not what I wanted/needed. I need multiple filament changes "within" a single layer. I do a lot of signage or designs that print face down and need multiple filaments to complete in that first layer.

So, like yourself I create multiple extruders, then assign each extruder to a single object in that first layer. That invokes the T0, T1 and so on to select the extruder, except with the firmware it then changes the T0, 1, 2, to an M600 filament change (not in the GCODE, in the firmware).

If you go main Kobra Max page here, scroll to the bottom and you'll see how I use the change within a single layer.

You might be able to work around the firmware if you edited the GCODE generated to remove all T0, T1, etc from the code. Not 100% sure that would work, but at least it wouldn't kick in the firmware with the Tx codes.

closed 2024-12-12 by wabbitguy

leonet99, 2024-12-12:
Thank you for your reply! I did suspect that may be the case, but I am fairly new to this and wasn't sure if this was intended, or a bug, and it's good to have it confirmed that the printer works as it supposed to with this FW, and that I didn't do something wrong either.
And your last suggestion is exactly what I did to get past this behavior: I edited the GCODE and removed all the tool changes from the code (well, I first changed all T1,T2... occurrences to T0 thinking that if the firmware doesn't see a new tool ID change from the initial T0, then it won't trigger the pause for it, but that didn't work and it performed exactly the same, so then I tried only leaving the first T0 and deleted the rest, which worked fine -no more pauses). So, I am also glad to know that I didn't do anything silly as there wasn't any other "smarter" choice available to me for it.
Well, no problem then, it's easy enough to do this, and I only need to do it whenever I need to run these types of tests anyway. This feature and the way you implemented (to change filament "within" a single layer) is actually very useful to work the way it is (and thanks for the usage tips, btw) so, now I am very glad to know how to and how not to use it. 
Thanks again for your response and for sharing the firmware! 👍

wabbitguy, 2024-12-13:
@leonet99 super to hear you got it sorted out! The first time the firmware see's the T0 it sets up some variables so it knows what to do when it comes across the T1, T2 and so GCODE's later on in the file. Hence, as you found leaving the first lil blighter in there won't hurt anything..LOL

The firmware I did largely because AnyCubic's implementation was brain dead. Frustrating me beyond belief so when they published the source code it didn't take me long to crawl in and start to modify it. I've done firmware for printers for almost a decade now so pretty familiar with Marlin. But Anycubics's code defies logic in far too many areas.

The motherboard they use, for all the clammer of 32 bit, has less memory than some ancient Arduino MEGA does. I.e. roughly 201K of memory. Trying to get a full Marlin code set back into the firmware and the motherboard was constantly running out of memory. Ugh.

Then, to top it off, for the second production run of the Kobra's Anycubic changed the motherboard to an even more limited and cheaper version. The existing firmware didn't like the new motherboard at all because of the "modifications" to the PCB they made. And then never released the firmware they used for the new board. Even though it's open source and they are supposed to. Sigh...

Any who, happy to share with the community and again, super glad you got it figure to make it work for you! Thanks for kind words!

Mel