wabbitguy/Kobra_Max #36: only self leveling half the build plate

soretoez, 2025-04-13:
Hi I changed my firmware because I wanted the baby steps, But on releveling the bed the print head only dabs about half of the x axis.
The y axis is fine and does the full front to back. I have the Tri-Gorilla_ProA V1.0.4 The only difference I have is the resistor has been moved from r66 to r65 could that be the issue? I bought the machine second hand so not sure what sort of life it had before I purchased it.

wabbitguy, 2025-05-03:
@soretoez sorry about the long delay. To be honest I'm not sure if that the resistor change would affect the levelling or not. I know the change had to be done if the firmware was setup to use linear advance in Marlin. In my own tests with linear advance enabled I couldn't find any advantage on the four Max's that I had at the time (I have none now).

If you use the LCD and move the head, does it cover the whole bed or only half of it?

closed 2025-05-03 by wabbitguy

wabbitguy, 2025-05-03:
@soretoez if you need a custom version of the firmware for the resistor change, I think I could do that for you. Have to look at the code and changes needed... let me know (and if the head moves like it's supposed to)...